claudeheintz/LXESP32DMX #1: Wifi Won't work with uart DMX

TheOnlyCaky, 2017-06-26:
Hi, Claude! GPIO16 and GPIO17 are used to connect to the embedded flash. I have not been able to get the wifi working when UART is connected to these pins. I use GPIO25 and GPIO26 for UART, and they work well for me!

claudeheintz, 2017-06-26:
On the module I have it is GPIO 9 and GPIO 10 that are connected to flash (These are mapped to UART1 RX and TX (function 5).  That's the reason that I selected UART2 to use for DMX.  I've updated the ESP32DMX driver to allow a custom pin to be passed in the startOutput and startInput methods.

According to the technical reference manual (pg. 39), you can map any function to any pin but the standard function mapping has improved high speed performance because it bypasses the GPIO matrix.  So, in theory, you could use UART1 on my module with different pins as I believe that the module's flash uses HSI not UART.

closed 2017-11-29 by claudeheintz

claudeheintz, 2019-01-19:
Thanks for your question.  I'm glad to respond to it.  However, this is off the topic of the original issue.  Please open a new issue and post your question there.

Danke für deine Frage. Ich bin froh, darauf zu antworten. Dies ist jedoch nicht das Thema der ursprünglichen Ausgabe. Bitte öffnen Sie eine neue Ausgabe und stellen Sie Ihre Frage dort auf.