claudeheintz/LXESP32DMX #3: Using UART0 instead of UART2

ririfonfon, 2018-02-14:
Hi Claude,
i was wondering how to use UART0 instead of UART2 (primary for receiving purpose)
since i use an oled screen equipped esp32 which already use pin 16 (it reset the screen..).

What i did:

- ESP32DMX.startInput(3); in my setup to force pin 3 as input (uart0 RX) instead of
the default pin 16 set @line 135 in LXESP32DMX.h

- LXHardwareSerial Serial2(0);  instead of (2) hopping to force use of Uart0 instead of Uart2
@line 23 in LXESP32DMX.cpp

but it doesn't seems to work. 
I have disabled Serial printing of course, and no USB are connected.

Any clue why it would not work ? Is there limitation to use UART0 ? 
Or maybe other hardcoded values that i should change ?

Thanks !

Best Regards,

claudeheintz, 2018-02-14:
You would need to alter the Arduino ESP core to reassign the uart pins.  This may help you figure out how to do that: https://hackaday.com/2017/08/17/secret-serial-port-for-arduinoesp32/
It may be easier and better to re-write the OLED library so that it uses a different pin.

claudeheintz, 2018-02-14:
Sorry, on further investigation, the ESP32/Arduino Hardware serial appears to always use the gpio_mux table and you can assign the pins when calling Serial.begin();  This does not seem to work on all pins, however.  And, the LXHardwareSerial.sendBreak method needs to account for the pin used.  I'll update the library for that after a little more testing.  I was able to use uart 2 as input through GPIO4 on my board with the input example simply by calling ESP32DMX.startInput(4); on line 112.

claudeheintz, 2018-02-15:
discussion of UART pins:  https://esp32.com/viewtopic.php?t=1033

claudeheintz, 2018-02-15:
The following is the results of tests using the inputTest and outputTest examples, modified to define the UART rx/tx pin used:

```
Pin  UART2RX UART0RX UART2TX UART0TX
IO15 No      Yes     Yes     Yes
IO2  No      Yes     Yes     Yes
IO0  No      No      Yes     Yes
IO4  Yes     Yes     Yes     Yes
IO16 Yes     Yes     Yes     Yes
IO17 Yes     Yes     Yes     Yes
IO5  Yes     Yes     Yes     Yes
IO18 Yes     Yes     Yes     Yes
IO19 No      Yes     Yes     Yes
IO21 No      Yes     Yes     Yes
IO22 No      Yes     Yes     Yes
IO23 No      Yes     Yes     Yes
IO34 Yes     Yes     No      No
IO35 Yes     Yes     No      No
IO32 Yes     Yes     Yes     Yes
IO33 Yes     Yes     Yes     Yes
IO25 No      Yes     Yes     Yes
IO26 Yes     Yes     Yes     Yes
IO27 No      Yes     Yes     Yes
IO14 No      Yes     Yes     Yes
IO12 No      Yes     Yes     Yes
IO13 No      Yes     Yes     Yes
```

closed 2019-01-09 by claudeheintz