claudeheintz/LXESP32DMX #8: Reset Problem when Wifi is used

Cubyte, 2019-03-06:


Cubyte, 2019-03-06:
when i connect to the WiFi network the esp32 device resets after about 40 seconds.
But if I use APmode  it's fine 
any ideas?

claudeheintz, 2019-03-09:
It is possible that the wifi router or the router is directing too much traffic to the ESP32 and it gets overloaded.  Or, there is an issue with multicast and the router.  On my home WiFi router, I cannot get multicast to work while the same code and hardware works fine on another network.  I haven't been able to track it down.  But, it is possible it is something like that where the ESP32 spends too long waiting and blocking on something and that overruns the WDT (which causes the ESP32 to reboot).  I'd try with another type of WiFi router to see if it is specific to one network or happens on both.