claudeheintz/LXESP32DMX #9: No input works

Shamanni, 2019-04-08:
Hi! I think, the problem maybe in differences between latest ESP32 core and your version of it. Can you help me with it?

claudeheintz, 2019-04-08:
I believe this is fixed with the latest commit which includes a copy of a replacement esp32-hal-uart.c file for the arduino-esp32 library that is compatible with the current commit of that library.

To use, find the esp32-hal-uart.c in the LXESP32DMX/extras/modified_hal-uart folder.  Move the modified file from LXESP32DMX/extras/modified_hal-uart/esp32-hal-uart.c   to   arduino-esp32/cores/esp32/esp32-hal-uart.c, replacing the existing file.

The differences between the standard esp32 file and the modified version are clearly marked with a comment that starts with "//begin mod".  The first of these defines a macro (#define USE_SLIP_FOR_BREAK_DETECT 1) that is used by a pre-compile directive (#if USE_SLIP_FOR_BREAK_DETECT) to switch the modification on and off.  To restore uart read behavior to the default, set line 38 to read #define USE_SLIP_FOR_BREAK_DETECT 0

The modification uses SLIP to signal when the DMX break is detected.  SLIP is a simple escape sequence technique that allows marking the end of a serial stream.

closed 2020-01-29 by claudeheintz

m-elias, 2020-02-02:
I'm also having an issue with receiving only. I only see one set of 0 values printed to the serial montior when using the inputTest example. I'm using a portable installation of the Arduino IDE so my directory structure looks different. I searched for my instance of esp32-hal-uart.c and replaced it but it didn't seem to help. The instuctions on your post April 8 2019 also appear to be different than on the readme which mentions the sketch folder.
arduino-esp32/cores/esp32/esp32-hal-uart.c
vs
[path to sketch folder]/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c
How can I confirm that it's using the correct file when compiling?