claudeheintz/LXESP32DMX #12: Removing the need to mod esp32-hal-uart

seymar, 2019-07-28:
First of all thank you for your great efforts in create this library! 👍 

However modding the code from the core kinda makes me scratch my head.
Wouldn't it be possible to just copy the functionality of [esp32-hal-uart](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.h) and changing it as required?

kbickham, 2019-08-01:
I have a wroom-32 properly configured...and can't get the hardware uart to output anything besides the serial monitor in the arduino ide (though usb).

claudeheintz, 2019-08-17:
The need to modify the esp32-hal-uart file is only for DMX input.  If you are using the library to transmit DMX, it is not necessary.  The reason that the mod is required is because RTOS places incoming bytes in a queue.  The only way to signal a break is to use a scheme like SLIP to encode the start/end of packets.  This is pretty specialized behavior.  So, its unlikely that it would be included in the esp32 library as a pull request.  Hence the need to modify the esp32-hal-uart file for the purposes of receiving DMX.

kbickham, 2019-08-17:
I would add, I have mine communicating via uart to a display and to a 328p.

closed 2020-01-29 by claudeheintz