claudeheintz/LXESP32DMX #13: Can't input high number DMX slots

macdroid53, 2019-08-21:
Hi,
I'm using the test code ```inputTest```

I have noticed two issues.

If I try to use a test_slot (i.e. test_slotA, test_slotB, etc.) if the slot value is greater than 13 that slot is not reported.
I have a logic analyzer on the input pin and it decodes all 512 DMX channels.

Any ideas what could be wrong?

Also, sometimes it never completes boot after the reset is pressed. (this could be crappy switch...but, I haven't noticed it before).

macdroid53, 2019-08-22:
After further testing, I discovered that if a slot specified in inputTest.ino here:
```
// the addresses of the slots to observe
int test_slotA = 10;
int test_slotB = 215;
int test_slotC = 13;
```
is zero when the program starts it is not reported.

If the slot is then changed it is reported.

I believe this is because in inputTest.ino the slot values are initialized to 0:
```
// the levels of those slots
uint8_t test_levelA = 0;
uint8_t test_levelB = 0;
uint8_t test_levelC = 0;
```

Then in the receiveCallback() it checks the received value against 0:
```
if ( test_levelA != ESP32DMX.getSlot(test_slotA) ) {
...
```
Thus the slot in question does not get reported until the slot changes from 0.

claudeheintz, 2019-08-23:
I think I just quoted Knuth's famous "premature optimization is the root of all evil" in response to another question.  The example does not necessarily need the check to see if the levels have changed.  It is there because something similar is necessary in the callback if whatever action is taken is expensive in terms of time--particularly something that can't be optimized like sending serial data to a long string of LEDs.  If something in the callback takes too long, it can prevent the input driver from running enough to keep up with the DMX stream.

With only 8 bits, there is no way to mark the stored level as being uninitialized.  Possible strategies could be to use a single flag so that all levels are set the first time the callback function is used.  Or, you could store each level in a 16bit integer allowing you to mark if the least significant byte has been set or not.  In either case, it adds to the complexity of the example, which is not intended as a final solution, but a starting point.

macdroid53, 2019-08-23:
I agree and have already tweaked my code to handle "first contact" situation. In fact with the flag you describe.
But, the example as written, would appear to do nothing if all selected slots happen to be zero when the program begins.
It took me a couple looks to notice the output statements would not execute in that case. And, since 2 of the 3 slots I happened to choose did have values other than zero, it appeared to be failing to report one of the chosen slots. Also, the DMX slots above 13 were all zero, further adding to my confusion.
I'm not dealing with LEDs, but triac controlled incandescent. So the pwm is sync'd to the line and I only need to write values to the PWM duty cycle register.

claudeheintz, 2019-08-23:
I will check the input example for slots higher than 13 as soon as I get a free hour or so.  As far as I know there is no problem and the last time I checked it worked.  But, I will verify and report back.

macdroid53, 2019-08-23:
Sorry, I was not clear. **There is no problem with slots above 13.**

It was coincidence that my DMX console was sending some value, other than zero, on 1 through 13, but, zero on 14 through 512. So, if I picked any slot above 13, it _appeared_ that it was receiving nothing above 13.

Once I realized the terminal output only happens when the slot changes, I saw what was happening.

I added:

```
bool initial_read = true;
```
Then changed the call back:
```
....
....
		if ( (test_levelC != ESP32DMX.getSlot(test_slotC)) || initial_read ) {
			test_levelC = ESP32DMX.getSlot(test_slotC);
			Serial.print(test_slotC);
			Serial.print(" => ");
			Serial.println(test_levelC);
			//gammaCorrectedWrite(led_channelC, test_levelC);
		}
		initial_read = false;
...
...
```

Now, on start up, even if the slot is zero it is reported.

closed 2020-03-27 by claudeheintz