claudeheintz/LXESP32DMX #14: PWM resolution and gpio 18 issue.

julienhogert, 2019-08-28:
Hey Claude ! 
Thanks for the amazing job you did. I used your library for dmx with an esp8266 and now I'm using it with an esp32 to control directly 16 leds 12v with mosfets. It works like a charm ! thanks a lot ! 
Now I have some questions. I´d like to increase the resolution of my pwm to get better results. - Would it be possible for it to work in a resolution other than 8bit? I can't make it work once I change the resolution and wanted to know if there was something that I could do. Could it maybe be related to a "clock" problem, and if so, is there a way to avoid it?
- Another issue is when I use the GPIO 18, it's not working anymore. Is there any reason for this?
- Last question: I think I will use an Arduino module for the rs485. There are 2 models that I could use:
this one :https://images.app.goo.gl/iyy9Nhgst8h6XZpy9
or this one : https://images.app.goo.gl/G9eBiMuqCqsqZQA58
The second one seems to have better protection for the Micro-controller, and I think I will use this second option. But I would like to know whether you have used the first option and whether you had encountered any problems with it (such as tension problems, grilling of the cards, etc.)

Finally, this is the code I use:

`#include <LXESP32DMX.h>
#include "esp_task_wdt.h"

#define DMX_DIRECTION_PIN 21
#define DMX_SERIAL_INPUT_PIN 16
#define RESOLUTION_PWM 8 

// -- VARS LEDS
int freq = 5000;
byte pins_leds[]={0,15,5,13,
                  12,4,17,14,    
                  21,22,19,23,                
                  27,25,26,13}; 
const byte nb_leds= sizeof(pins_leds)/sizeof(pins_leds[0]);
const int pwm_steps = pow(2,RESOLUTION_PWM);  
// -- VARS DMX
bool received_dmx=0;
int dmx_add = 1;

// ------------------ DMX RECEIVE FUNCTION
void receiveCallback(int slots) {
  if ( slots ) {
    received_dmx=1;
    xSemaphoreTake( ESP32DMX.lxDataLock, portMAX_DELAY );
    for(int i=0;i<nb_leds;i++){
       ledcWrite(i, ESP32DMX.getSlot(dmx_add+i));
    }
    xSemaphoreGive( ESP32DMX.lxDataLock );
  }
}

// ------------------ SETUP
void setup() {
  // ----- LEDS
  for(int i=0;i<nb_leds;i++){
      ledcSetup(i, freq, RESOLUTION_PWM);
      ledcAttachPin(pins_leds[i], i);
  }
  ledcAttachPin(2, 0); // BUILTIN LED
  
  // ----- DMX
  pinMode(DMX_DIRECTION_PIN, OUTPUT);
  pinMode(DMX_SERIAL_INPUT_PIN, INPUT);
  digitalWrite(DMX_DIRECTION_PIN, LOW);
  
  ESP32DMX.startInput(DMX_SERIAL_INPUT_PIN);
  ESP32DMX.setDataReceivedCallback(receiveCallback);
}
// ------------------ SETUP
void loop() {
if(!received_dmx){
  for (int pwm = 0; pwm <= pwm_steps; pwm+=2) {
    for(int i=0;i<nb_leds;i++){ 
        ledcWrite(i, pwm); 
      }
    delay(7);
  }
  for (int pwm = pwm_steps; pwm >= 0; pwm-=2) {
    for(int i=0;i<nb_leds;i++){
        ledcWrite(i, pwm);
     }
    delay(7);
  }
}
}`  

Again, thanks for sharing your work.
Julien

claudeheintz, 2019-08-29:
I will have to take a closer look at the PWM resolution to see how it works.  What happens when you try this?  Is there no output or unexpected output?

Of the boards, the one that will work with the library is probably the first one.  The second one does appear to have more protection.  However, I don't see how the serial direction is controlled.  You's want to be sure you can switch between input and output.  The lack of protection depends on how you are powering the boards.  If its attached to a computer and there's a transient on the DMX or a fault in a fixture that puts line voltage on the DMX, it could be costly and take out your computer.  The risk in your situation would potentially not cost the computer if you use a cheap power supply instead, but it could take out the ESP32 and LEDs...  Depends on the situation what the risk of line voltage on the DMX really is...  Like any insurance, you have to balance the cost with the potential consequences.  I absolutely go with optical isolation if I'm connecting to my computer and DMX.

Finally, I would test pin 18 on your board with something else, a simple sketch that just proves if it works or not.  It could be that specific board.  With the ESP32, the chip pins are not necessarily the same as the arduino pins and that's something to keep in mind when you look at the documentation and see the functions available for a pin.  (I also don't see what you are trying to use pin 18 for in the above sketch)

julienhogert, 2019-09-10:
Hello Claude !
I'm continuing my investigations. Finally, I was wrong about the PWM resolution. It was a mess between my "loop test" and the dmx received. So I was able to use dmx and 12 bit resolution without problem. But beyond 12, I has some Flikering problem, but probably because of the frequency. About the pin 18, I didn't check yet.  And about the 2 shield. I don't why, the one was working at a moment (few month ago), but now I don't find a way to make it works and there no documentation about it. So I will use the first one even if it looks less protected.
When I will have more infos, I will send you.
Thanks,
Julien

macdroid53, 2019-11-17:
Just curious. DMX is an 8 bit protocol. So it can represent a max of 255 discrete levels. So are you mapping the 12 bit value somehow? Or are you using 2 channels to get 12 bits?

julienhogert, 2019-11-17:
No, I only use one single channel of DMX. but then I make a logarithmic scale to 12 bits. Like this, the led don't stop with very low values, but fade out slowly. I use this lookuptable :
`const uint16_t CIEL_8_12[] PROGMEM = {
  0, 2, 4, 5, 7, 9, 11, 12, 14, 16, 18, 20, 21, 23, 25, 27, 28, 30, 32, 34, 36, 37, 39, 41, 43, 45, 47, 49, 52, 54, 56, 59, 61, 64, 66, 69, 72, 75, 77, 80, 83, 87, 90, 93, 97, 100, 103, 107, 111, 115, 118, 122, 126,
  131, 135, 139, 144, 148, 153, 157, 162, 167, 172, 177, 182, 187, 193, 198, 204, 209, 215, 221, 227, 233, 239, 246, 252, 259, 265, 272, 279, 286, 293, 300, 308, 315, 323, 330, 338, 346, 354, 362, 371,
  379, 388, 396, 405, 414, 423, 432, 442, 451, 461, 471, 480, 490, 501, 511, 521, 532, 543, 554, 565, 576, 587, 599, 610, 622, 634, 646, 658, 670, 683, 696, 708, 721, 734, 748, 761, 775, 789, 802, 817,
  831, 845, 860, 875, 890, 905, 920, 935, 951, 967, 983, 999, 1015, 1032, 1048, 1065, 1082, 1099, 1117, 1134, 1152, 1170, 1188, 1206, 1225, 1243, 1262, 1281, 1301, 1320, 1340, 1359, 1379, 1400, 1420,
  1441, 1461, 1482, 1504, 1525, 1547, 1568, 1590, 1613, 1635, 1658, 1681, 1704, 1727, 1750, 1774, 1798, 1822, 1846, 1871, 1896, 1921, 1946, 1971, 1997, 2023, 2049, 2075, 2101, 2128, 2155, 2182, 2210,
  2237, 2265, 2293, 2322, 2350, 2379, 2408, 2437, 2467, 2497, 2527, 2557, 2587, 2618, 2649, 2680, 2712, 2743, 2775, 2807, 2840, 2872, 2905, 2938, 2972, 3006, 3039, 3074, 3108, 3143, 3178, 3213, 3248,
  3284, 3320, 3356, 3393, 3430, 3467, 3504, 3542, 3579, 3617, 3656, 3694, 3733, 3773, 3812, 3852, 3892, 3932, 3973, 4013, 4055, 4095 };`

And you're right: some programs use 2 dmx channels to have a 16bits resolution.
Claude: Finally, the second shield works well too.