claudeheintz/LXESP32DMX #15: Crashing regularly

Harkunwar, 2019-08-30:
Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c:1438 (vTaskDelay)- assert failed!
abort() was called at PC 0x400871e1 on core 1

Backtrace: 0x40089d4c:0x3ffb3f60 0x40089f79:0x3ffb3f80 0x400871e1:0x3ffb3fa0 0x400d1279:0x3ffb3fc0 0x400885c9:0x3ffb3fe0

With the ESP Exception Decoder:

0x40089d4c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 707
0x40089f79: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 707
0x400871e1: vTaskDelay at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 3537
0x400d1279: receiveDMX(void*) at LXESP32DMX1.cpp line ?
0x400885c9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 355 (discriminator 1

claudeheintz, 2019-08-31:
Is this being used with the Arduino esp32 board package or something else?  I don't see the source file freertos/tasks.c to be able to look at the assertion that is failing.  If you are not using the esp32 Arduino library, you could look at how you are building freertos to see if there is something you need to configure.

from freertos/task.h:
```
 * INCLUDE_vTaskDelayUntil must be defined as 1 for this function to be available.
 * See the configuration section for more information.
```

Harkunwar, 2019-08-31:
I've been using Arduino with an ESP32 on PlatformIO. I was able to receive DMX normally but it was crashing in 10 minutes or so. I change the priority of your DMX receive task from 1 to 6 and it hasn't crashed in a few hrs. I have to check it further if it's perfectly fine though. 
Personally I tried looking for that assertion too and I couldn't find anything. Maybe that's because PlatformIO uses precompiled libraries and using the latest from the git might solve the issue too?

Harkunwar, 2019-08-31:
Also wanna say thanks for the amazing library. I've been mostly using it for output on several devices and has worked great always!

closed 2019-09-07 by Harkunwar