claudeheintz/LXESP32DMX #21: Compilation error witch patched hal

Maka-120774, 2020-02-17:
sketch "input test" compiles with errors:

/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-uart.c: In function 'uartFlush':
/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-uart.c:430:5: error: implicit declaration of function 'uartFlushTxOnly' [-Werror=implicit-function-declaration]
     uartFlushTxOnly(uart,false);
     ^
/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-uart.c: At top level:
/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-uart.c:433:6: warning: conflicting types for 'uartFlushTxOnly'
 void uartFlushTxOnly(uart_t* uart, bool txOnly)
      ^
/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/esp32-hal-uart.c:430:5: note: previous implicit declaration of 'uartFlushTxOnly' was here
     uartFlushTxOnly(uart,false);
     ^

Maka-120774, 2020-02-21:
I did this: 
I put the "void uartFlushTxOnly (uart_t * uart, bool txOnly) {...} "function above the "void uartFlush (uart_t * uart) {...}" function and everything began to compile without problems.

claudeheintz, 2020-03-02:
There are now two replacement esp32-hal-uart.c files, one for the board package from github and one if esp32 is installed using boards manager in the Arduino IDE.  The difference has nothing to do with the modifications for DMX input.  The changes are identical, but the starting file is different.

closed 2020-03-02 by claudeheintz