claudeheintz/LXESP32DMX #25: modified esp32-hal-uart uartEnd Argument Count

modellking, 2020-11-04:
HardwareSerial uses uartEnd with 3 Arguments, the implementatino infers them from the uart object. So 2 Versions of uartEnd are used.

Hacky Solution: Change the Signature in the Header and cpp to
`void uartEnd(uart_t* uart, ...)`

Did the versions get out of sync? Is this Library still maintained?

modellking, 2020-11-06:
btw. I do not receive anything after fixing above issue... How sensitive is the break detection? (I'm using a commercially available but cheap DMX-Controller to send DMX (an Arduino with DXMSerial is receiving everything it sends thou))

claudeheintz, 2020-12-29:
The change in uartEnd appears to be in the ESP32-Arduino Library on github (v1.0.5) rather than the one accessed through Boards Manager.  There are now multiple replacements for  esp32-hal-uart.c in the extras folder for various versions of ESP32-Arduino.  The modifications are clearly marked with "//begin mod" and "//end mod" and are the same in all of them.

If you need a compatible esp32-hal-uart.c file and one of these doesn't work, start with the esp32-hal-uart.c for the version you are using and modify it, rather than trying to fix one of the ones from the extras folder up so it compiles.   You can take any esp32-hal-uart.c and add the #defines at the top and insert the SLIP/Break Detect code into static void IRAM_ATTR _uart_isr(void *arg).  The only modifications are in those two places.

This is only necessary for DMX input.  And, quite frankly, there are a lot of better microcontrollers to choose if you are working on a project that requires DMX input.  For example, ESP8266 is much more stable and reliable with DMXInput than ESP32 is.  The reason for this may very well be the break detect.  There have been lots of problems related to partially read or out of sync DMX data with the ESP32.

cgiles, 2021-08-06:
Hello I use platformio+vscode and I got a similar error :
`lib\LXESP32DMX\src\LXHardwareSerial.cpp: In member function 'void LXHardwareSerial::end()':
lib\LXESP32DMX\src\LXHardwareSerial.cpp:229:46: error: too many arguments to function 'void uartEnd(uart_t*)'
     uartEnd(_uart, _rx_gpio_pin, _tx_gpio_pin);  //Arduino esp32 1.0.5
                                              ^
In file included from D:\Users\User\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal.h:52:0,
                 from D:\Users\User\.platformio\packages\framework-arduinoespressif32\cores\esp32/HardwareSerial.h:51,
                 from lib\LXESP32DMX\src\LXHardwareSerial.h:17,
                 from lib\LXESP32DMX\src\LXHardwareSerial.cpp:16:
D:\Users\User\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-uart.h:56:6: note: declared here
 void uartEnd(uart_t* uart);
      ^
*** [.pio\build\esp32doit-devkit-v1\libca8\LXESP32DMX\LXHardwareSerial.cpp.o] Error 1

claudeheintz, 2021-08-06:
The modified esp32-hal-uart.c file included with the libaray needs to be used.  This is currently compatible with ESP32/Arduino 1.0.5.  Version 1.0.6 and 2.0.0 compatibility is unknown at this time.

Break detection (used for DMX input by this library) is not well documented by espressif.

cgiles, 2021-08-07:
Thanks for your answer, I will look for modify the file included with this package.