claudeheintz/LXESP32DMX #28: Can't get any DMX input working

fastfourier666, 2021-02-26:
Hi Claude,

I'm trying to get some DMX input working, but with no success. I have Arduino 1.8.13 and ESP32 1.0.5.

I have tried downloading the ESP32 core both via the boards manager and cloning from github. The inputTest example does compile, but no DMX is shown in the serial monitor. I have tried with and without the replacement HAL files but no luck.

I did go through the closed issues regarding no input, and none of the solutions appear to apply to me. I'm using a Enttec DMX USB Pro to send the DMX and it is sending all 512 channels. The same setup works with the ESP32-DMX-RX library (https://github.com/luksal/ESP32-DMX-RX) so I'm 99% sure my hardware is good.

On line 27 of inputTest.ino I noticed:

`#define DMX_SERIAL_INPUT_PIN 34`
should that not be 16? I tried changing it but no dice.

Thank you for any help you can give!

IS512, 2021-02-26:
It sounds like my issue 

https://github.com/claudeheintz/LXESP32DMX/issues/27

If it does work for me it's extremely intermittent or incorrect data. 

Yeah from memory pin 16 is correct it's either (16 or 17). Everything works fine with the other library for me too.

fastfourier666, 2021-03-01:
I wonder, could I make it work by rolling back to a previous version of either this library or the ESP32 core?

claudeheintz, 2021-03-01:
The library uses the ESP32's break detection feature to identify the start/stop of each DMX packet in the stream.  I'm not sure how the other library functions.  It may do its own timing rather than relying on the ESP32's UART I/O modules.

Because of RTOS, it is extremely difficult to diagnose serial timing issues on the ESP32.  You can't just put a scope on a pin and observe when an interrupt is triggered the way you can with other microcontrollers.  The ESP8266 is much simpler and more reliable for DMX input, for example.  Unless there's a strong reason for choosing ESP32, I'd go with either ESP8266 or another processor.  There are plenty to choose from that are as fast as the ESP32.

The last time I did a test of this library, I used 1.0.5 rc and Arduino 1.8.13.  DMX input was functioning at that time.  But, I did not specifically try an ENTTEC DMX USB Pro.  They do allow you to adjust the break timing and you can try playing around with that to see if it makes a difference.  The last time I checked, the ESP32 SDK did not give any options for the break detect timing.

shridattdudhat, 2021-09-19:
Hello,
I am facing same issue, No data in inputTest Sketch,
I am using Arduino 1.8.16, and tried ESP32 core 1.0.5 and 1.0.6 but no results. Also replaced uart-hal files one by one still not working. there are many files in extras folder i dont know which to choose. 

any help will be appreciated ! 

[image]

shridattdudhat, 2021-09-20:
Hello, Finally it gets working!!!
Used ESP32 core version 2.0.0rc1, Arduino 1.8.16 & Board DOIT ESP32 Devkit1,
[image]

With ESP32 Dev Module not working choose DOIT ESP32 Devkit!

claudeheintz, 2021-09-20:
Verified the input example today with Arduino 1.8.13 and Arduino ESP (Boards Manager) version 2.0.0rc1.  DMX input was from a couple of sources including an ENTTEC DMX USB Pro mk2.