claudeheintz/LXESP32DMX #34: ESP32 issues - solved with a delay of 25ms . But not without

julienhogert, 2022-10-19:
Hello Claude, 
Thanks again for the great work you did by publishing your libraries. I'm working on a project with an esp32. (the board dev kit v1) combining DMX (rs485) / OSC and leds (with fastled).
For a time I wasn't able to make it works properly, with or without Wifi. I had some zeros, gaps in the DMX, some time shifts between differents slots.
Then I came back to your examples, and with this small delay(25) in the loop, it make the all thing working again. Even if the fastled library needs to call an update the most often possible, It looks there is no problem with leds, wifi, osc and other functions, that's nice !
But for the same project I need to control a brushless motor. Then I use the library Simplefoc which doesn't allow any delay. I tried to take out the attachment of the callback function.  //ESP32DMX.setDataReceivedCallback(dmx_to_bldc);

To put a non-blocking call in the loop : 
  if(millis()> millis_dmx+25){
    millis_dmx=millis();
    dmx_to_bldc();
  }
and keeping the xSemaphoreTake( ESP32DMX.lxDataLock, portMAX_DELAY ) in my function dmx_to_bldc();
But it didn't work. I have some erratic values.  
So, I wanted to ask you if there is any way to make the library to work without this delay? 
my arduino version is 2.0.0, the esp32 board library is 2.0.3, and without any modification of the hal file, as you mentioned that is no more necessary.
In other examples, you use vTaskDelay(25). Is there any difference between this solution and the delay(25)? (I didn't try yet.
Thanks for your help,

Julien

julienhogert, 2022-11-10:
Any suggestions? Thanks, Julien

claudeheintz, 2022-11-10:
vTaskDelay is an RTOS call which allows the scheduler to run other tasks.

delay() is an Arduino function.  I don't know how it is implemented for esp32.  If it blocks and no other task can run, then no, it won't work.