claudeheintz/LXESP32DMX #35: Problems reading DMX depending on packet size

mansj, 2022-11-23:
I've got a couple of different DMX controllers, and only one of them work with this library when I attemp to read from it. I've used the sniffer code from https://github.com/someweisguy/esp_dmx to look at what's happening, and I can see a few differences.

With the board _not_ working, I see this in the DMX Sniffer output: 

`Start Code: 0x00, Size: 13, Slot 1: 0x98, Break: -1us, MAB: -1us Packet: 20.00ms`

With the board that _is_ working, I see this:

`Start Code: 0x00, Size: 33, Slot 1: 0xFF, Break: -1us, MAB: -1us Packet: 9.52ms`

The data written is size of the packet, the slot (DMX channel) and the packet duration in ms. I suspect the main reason why the boards are not working in this library is because of their different packet size? 

The DMX Sniffer code is here: https://github.com/someweisguy/esp_dmx/tree/release/v2.0/examples/DMXSniffer

jatatech, 2022-11-23:
Hi @mansj ,

This will be due to the default value of
#define DMX_MIN_SLOTS 24

This requires a minimum of 24 slots/channels.

mansj, 2022-11-23:
Thanks for the response @jatatech - I tried defining `#define DMX_MIN_SLOTS 12`, since my controller has 12 DMX channels, but there is no difference. Am I missing something here, would I need to put this definition somewhere else than at the top of my code?

mansj, 2022-11-23:
Confirmed, when I change the `DMX_MIN_SLOTS´ value to 12 in the source code (LXESP32DMX.h) it works as it should. Thanks for the help @jatatech !

closed 2022-11-23 by mansj